AgustinLorenzo/openwrt #3: AX3600 Trying to pair a Tuya Device crashes radios

Ameb, 2023-07-07:
### Describe the bug

I have a set of tuya smart plugs which will crash all the wifi networks on my AX3600 when starting the pairing procedure (in both Smart Life and Nooie app).

I will add more info as needed, contacting you on Telegram.

```
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: fatal error received:
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] QC Image Version: QC_IMAGE_VERSION_STRING=WLAN.HK.2.9.0.1-01385-QCAHKSWPL_SILICONZ-1
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] Image Variant : IMAGE_VARIANT_STRING=8074.wlanfw.eval_v2Q
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398]
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] wal_peer_control.c:2870 Assertion is_graceful_to_handle failedparam0 :zero, param1 :zero, param2 :zero.
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] Thread ID      : 0x00000060  Thread name    : WLAN RT1  Process ID     : 0
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] Register:
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] SP : 0x4bfd5938
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] FP : 0x4bfd5940
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] PC : 0x4b1080c4
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] SSR : 0x00000008
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] BADVA : 0x00020000
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] LR : 0x4b107860
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398]
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] Stack Dump
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] from : 0x4bfd5938
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398] to   : 0x4bfd61b0
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.007398]
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.056620] remoteproc remoteproc0: crash detected in cd00000.q6v5_wcss: type fatal error
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.079234] remoteproc remoteproc0: handling crash #1 in cd00000.q6v5_wcss
Fri Jul  7 12:38:13 2023 kern.err kernel: [  867.086882] remoteproc remoteproc0: recovering cd00000.q6v5_wcss
Fri Jul  7 12:38:13 2023 kern.info kernel: [  867.120538] remoteproc remoteproc0: stopped remote processor cd00000.q6v5_wcss
Fri Jul  7 12:38:13 2023 kern.warn kernel: [  867.129993] ath11k c000000.wifi: failed to find peer fc:67:1f:xx:xx:xx on vdev 2 after creation
Fri Jul  7 12:38:13 2023 kern.warn kernel: [  867.130050] ath11k c000000.wifi: failed to find peer vdev_id 2 addr fc:67:1f:xx:xx:xx in delete
Fri Jul  7 12:38:13 2023 kern.warn kernel: [  867.137555] ath11k c000000.wifi: failed peer fc:67:1f:xx:xx:xx delete vdev_id 2 fallback ret -22
Fri Jul  7 12:38:13 2023 kern.warn kernel: [  867.146225] ath11k c000000.wifi: Failed to add peer: fc:67:1f:xx:xx:xx for VDEV: 2
Fri Jul  7 12:38:13 2023 kern.warn kernel: [  867.155275] ath11k c000000.wifi: Failed to add station: fc:67:1f:xx:xx:xx for VDEV: 2
Fri Jul  7 12:38:13 2023 daemon.notice hostapd: phy2-ap1: STA fc:67:1f:xx:xx:xx IEEE 802.11: Could not add STA to kernel driver
Fri Jul  7 12:38:13 2023 daemon.notice hostapd: phy2-ap0: STA fc:67:1f:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 12:38:13 2023 kern.warn kernel: [  867.371999] ath11k c000000.wifi: failed to send WMI_PEER_DELETE cmd
Fri Jul  7 12:38:13 2023 kern.warn kernel: [  867.372049] ath11k c000000.wifi: failed to delete peer vdev_id 0 addr fc:67:1f:xx:xx:xx ret -108
Fri Jul  7 12:38:13 2023 kern.warn kernel: [  867.377112] ath11k c000000.wifi: Failed to delete peer: fc:67:1f:xx:xx:xx for VDEV: 0
Fri Jul  7 12:38:13 2023 kern.warn kernel: [  867.386128] ath11k c000000.wifi: Found peer entry 28:d1:27:4c:a7:6c n vdev 0 after it was supposedly removed
Fri Jul  7 12:38:13 2023 kern.warn kernel: [  867.393916] ------------[ cut here ]------------
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.403732] WARNING: CPU: 1 PID: 2299 at sta_set_sinfo+0xb80/0xbe0 [mac80211]
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.408337] Modules linked in: ecm xt_connlimit nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nf_conncount iptable_nat ath11k_ahb ath11k ath10k_pci ath10k_core ath xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat_rtsp nf_nat nf_flow_table nf_conntrack_rtsp nf_conntrack mac80211 iptable_mangle iptable_filter ipt_REJECT ip_tables cfg80211 xt_time xt_tcpudp xt_recent xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG x_tables wireguard tcp_bbr qrtr_smd qrtr qmi_helpers ppp_async nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv4 libcrc32c libchacha20poly1305 crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.408531]  cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact qca_nss_tunipip6 qca_nss_tun6rd qca_nss_cfi_cryptoapi qca_nss_crypto qca_nss_qdisc qca_nss_vxlanmgr qca_nss_pptp pptp qca_nss_pppoe pppoe pppox qca_nss_map_t qca_nss_lag_mgr qca_nss_l2tpv2 ppp_generic slhc qca_nss_gre qca_nss_bridge_mgr qca_nss_vlan qca_mcs bonding ip6_gre ip_gre gre ifb nat46 nf_defrag_ipv6 sit qca_nss_drv ip6_tunnel tunnel6 tunnel4 tun vxlan seqiv michael_mic leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom qca_nss_dp qca_ssdk gpio_button_hotplug aquantia hwmon crc32c_generic
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.530544] CPU: 1 PID: 2299 Comm: hostapd Not tainted 5.15.114 #0
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.552779] Hardware name: Xiaomi AX3600 (DT)
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.558939] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.563369] pc : sta_set_sinfo+0xb80/0xbe0 [mac80211]
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.570137] lr : sta_set_sinfo+0xb7c/0xbe0 [mac80211]
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.575346] sp : ffffffc009b83850
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.580376] x29: ffffffc009b83850 x28: ffffff800622a580 x27: ffffffc009b83dd0
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.583683] x26: 0000000000000000 x25: ffffffc008c02500 x24: ffffffc000ce0000
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.590801] x23: ffffffc009b83a48 x22: ffffff80062e48c0 x21: ffffff8003622aa8
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.597920] x20: ffffff80079d0880 x19: ffffff8003622000 x18: ffffffc008afe5e8
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.605038] x17: 302076656476206e x16: 2063363a37613a63 x15: 00000000000005f4
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.612155] x14: 00000000000001fc x13: 0000000000000001 x12: 0000000000000000
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.619273] x11: 0000000000000001 x10: 0000000000000118 x9 : 0000000000000400
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.626391] x8 : ffffff801fea9b40 x7 : 0000000000008ea8 x6 : 0000000000000015
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.633509] x5 : ffffff800622a640 x4 : 0000000000000000 x3 : ffffff800622a580
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.640627] x2 : 0000000000000000 x1 : ffffff800622a580 x0 : 00000000ffffff94
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.647745] Call trace:
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.654855]  sta_set_sinfo+0xb80/0xbe0 [mac80211]
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.657117]  sta_info_destroy_addr_bss+0x50/0x7c [mac80211]
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.661978]  ieee80211_nan_func_match+0x1b68/0x1e10 [mac80211]
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.667362]  cfg80211_check_station_change+0x1408/0x4c80 [cfg80211]
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.673264]  genl_family_rcv_msg_doit+0xb8/0x11c
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.679424]  genl_rcv_msg+0xd4/0x1cc
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.684284]  netlink_rcv_skb+0x5c/0x130
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.687843]  genl_rcv+0x38/0x50
Fri Jul  7 12:38:14 2023 kern.warn kernel: [  867.691401]  netlink_unicast+0x184/0x280
[... 23 lines cut ...]
[... cut ...]
Fri Jul  7 12:38:19 2023 kern.warn kernel: [  872.575093] ath11k c000000.wifi: failed to set BA BUFFER SIZE 256 for vdev: 0
Fri Jul  7 12:38:19 2023 kern.warn kernel: [  872.581776] ath11k c000000.wifi: failed to send WMI_VDEV_SET_PARAM_CMDID
Fri Jul  7 12:38:25 2023 kern.warn kernel: [  878.590487] ath11k_warn: 157 callbacks suppressed
```

### OpenWrt version

r0-2731da5

### OpenWrt target/subtarget

ipq807x/generic

### Device

Xiaomi AX3600

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

AgustinLorenzo, 2023-09-23:
Hi @Ameb 

Try in the latest firmware.

Regards, Agustin

closed 2023-09-23 by AgustinLorenzo

Ameb, 2023-09-28:
Still happening on `OpenWrt SNAPSHOT r0-ded870a / LuCI Master git-23.268.26539-2a09ea0`

```
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: fatal error received:
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] QC Image Version: QC_IMAGE_VERSION_STRING=WLAN.HK.2.9.0.1-01385-QCAHKSWPL_SILICONZ-1
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] Image Variant : IMAGE_VARIANT_STRING=8074.wlanfw.eval_v2Q
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085]
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] wal_peer_control.c:2870 Assertion is_graceful_to_handle failedparam0 :zero, param1 :zero, param2 :zero.
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] Thread ID      : 0x00000060  Thread name    : WLAN RT1  Process ID     : 0
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] Register:
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] SP : 0x4bfd5938
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] FP : 0x4bfd5940
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] PC : 0x4b1080c4
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] SSR : 0x00000008
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] BADVA : 0x00020000
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] LR : 0x4b107860
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085]
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] Stack Dump
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] from : 0x4bfd5938
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085] to   : 0x4bfd61b0
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.810085]
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.859083] remoteproc remoteproc0: crash detected in cd00000.q6v5_wcss: type fatal error
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.881319] remoteproc remoteproc0: handling crash #1 in cd00000.q6v5_wcss
Fri Sep 29 00:08:38 2023 kern.err kernel: [  431.889589] remoteproc remoteproc0: recovering cd00000.q6v5_wcss
Fri Sep 29 00:08:38 2023 kern.info kernel: [  431.922332] remoteproc remoteproc0: stopped remote processor cd00000.q6v5_wcss
Fri Sep 29 00:08:38 2023 kern.warn kernel: [  431.922640] ath11k c000000.wifi: failed to find peer fc:67:1f:xx:xx:xx on vdev 1 after creation
Fri Sep 29 00:08:38 2023 kern.warn kernel: [  431.928592] ath11k c000000.wifi: failed to find peer vdev_id 1 addr fc:67:1f:xx:xx:xx in delete
Fri Sep 29 00:08:38 2023 kern.warn kernel: [  431.937415] ath11k c000000.wifi: failed peer fc:67:1f:xx:xx:xx delete vdev_id 1 fallback ret -22
Fri Sep 29 00:08:38 2023 kern.warn kernel: [  431.945765] ath11k c000000.wifi: Failed to add peer: fc:67:1f:xx:xx:xx for VDEV: 1
Fri Sep 29 00:08:38 2023 kern.warn kernel: [  431.954777] ath11k c000000.wifi: Failed to add station: fc:67:1f:xx:xx:xx for VDEV: 1
Fri Sep 29 00:08:38 2023 kern.warn kernel: [  431.962188] ath11k c000000.wifi: failed to send WMI_REQUEST_STATS cmd
Fri Sep 29 00:08:38 2023 kern.warn kernel: [  431.970048] ath11k c000000.wifi: could not request fw stats (-108)
Fri Sep 29 00:08:38 2023 kern.warn kernel: [  431.976482] ath11k c000000.wifi: failed to request fw pdev stats: -108
Fri Sep 29 00:08:38 2023 kern.warn kernel: [  431.982571] ath11k c000000.wifi: failed to send WMI_REQUEST_STATS cmd
Fri Sep 29 00:08:38 2023 kern.warn kernel: [  431.989048] ath11k c000000.wifi: could not request fw stats (-108)
Fri Sep 29 00:08:38 2023 daemon.notice hostapd: phy2-ap0: STA fc:67:1f:xx:xx:xx IEEE 802.11: Could not add STA to kernel driver
Fri Sep 29 00:08:38 2023 daemon.notice hostapd: phy2-ap1: STA fc:67:1f:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Fri Sep 29 00:08:39 2023 daemon.warn omcproxy: mrib_mld_source: failed to detect local source address on 13
Fri Sep 29 00:08:43 2023 kern.err kernel: [  437.192070] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: start timed out
Fri Sep 29 00:08:43 2023 kern.err kernel: [  437.192122] remoteproc remoteproc0: can't start rproc cd00000.q6v5_wcss: -110
Fri Sep 29 00:08:43 2023 kern.warn kernel: [  437.272067] ath11k_warn: 4 callbacks suppressed
Fri Sep 29 00:08:43 2023 kern.warn kernel: [  437.272088] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 1
Fri Sep 29 00:08:43 2023 kern.warn kernel: [  437.275493] ath11k c000000.wifi: failed to send WMI_PEER_DELETE cmd
Fri Sep 29 00:08:43 2023 kern.warn kernel: [  437.283446] ath11k c000000.wifi: failed to delete peer vdev_id 2 addr fc:67:1f:xx:xx:xx ret -108
Fri Sep 29 00:08:43 2023 kern.warn kernel: [  437.289409] ath11k c000000.wifi: Failed to delete peer: fc:67:1f:xx:xx:xx for VDEV: 2
Fri Sep 29 00:08:43 2023 kern.warn kernel: [  437.298451] ath11k c000000.wifi: Found peer entry 2a:d1:27:xx:xx:xx n vdev 2 after it was supposedly removed
Fri Sep 29 00:08:44 2023 kern.warn kernel: [  437.306244] ------------[ cut here ]------------
Fri Sep 29 00:08:44 2023 kern.warn kernel: [  437.316049] WARNING: CPU: 0 PID: 2367 at sta_set_sinfo+0xcd0/0xd14 [mac80211]
Fri Sep 29 00:08:44 2023 kern.warn kernel: [  437.320654] Modules linked in: ecm xt_connlimit nft_fib_inet nf_flow_table_inet nf_conncount iptable_nat ath11k_ahb ath11k ath10k_pci ath10k_core ath xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE wireguard nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nf [... cut ...]
Fri Sep 29 00:08:44 2023 kern.warn kernel: [  437.633149]  invoke_syscall.constprop.0+0x5c/0x104
Fri Sep 29 00:08:44 2023 kern.warn kernel: [  437.637056]  do_el0_svc+0x58/0x16c
Fri Sep 29 00:08:44 2023 kern.warn kernel: [  437.641828]  el0_svc+0x18/0x54
Fri Sep 29 00:08:44 2023 kern.warn kernel: [  437.645212]  el0t_64_sync_handler+0x114/0x120
Fri Sep 29 00:08:44 2023 kern.warn kernel: [  437.648253]  el0t_64_sync+0x174/0x178
Fri Sep 29 00:08:44 2023 kern.warn kernel: [  437.652681] ---[ end trace 0000000000000000 ]---
```

Ameb, 2023-12-09:
Reported upstream

https://github.com/openwrt/openwrt/issues/14181