AgustinLorenzo/openwrt #4: AX3600: High latency and packet loss between wifi and wired clients

Branyac, 2023-08-21:
### Describe the bug

I'm having high latency and some packet loss problems between wifi and wired clients after installing the latest release [Updated prebuilt images 2023-08-21-0921](https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-2023-08-21-0921) in a Xiaomi AX3600 router.

The router is configured as Access Point. The firewall and the DHCP and DNS server are disabled. The internet connection is attached to port eth1.

The problem dissapears when I rollback to the previous version [Updated prebuilt images 2023-06-12-2358](https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-2023-06-12-2358)

### OpenWrt version

Updated prebuilt images 2023-08-21-0921

### OpenWrt target/subtarget

ipq807x/generic

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

1. Install [Updated prebuilt images 2023-06-12-2358](https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-2023-06-12-2358)

2. Configure Xiaomi AX3600 as Access Point (Change IP address and disable dnsmasq and firewall) Connect the router with Internet access to the port eth1

3. Update softwate to [Updated prebuilt images 2023-08-21-0921](https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-2023-08-21-0921) without reseting the router settings.

4. Connect to Wifi and ping to any wired client.

### Actual behaviour

IPv4 and IPv6 ping times are too high between wifi clients and wired or Internet clients. Sometimes packet loss happens.

### Additional info

Note: The following pings are made from a computer connected to the router by wifi to computers connected by cable.

```
root@OpenWrt:~# ping 192.168.1.1 -c 4
PING 192.168.1.1 (192.168.1.1): 56 data bytes
64 bytes from 192.168.1.1: seq=0 ttl=64 time=8515.045 ms
64 bytes from 192.168.1.1: seq=1 ttl=64 time=7514.928 ms
64 bytes from 192.168.1.1: seq=2 ttl=64 time=6514.804 ms
64 bytes from 192.168.1.1: seq=3 ttl=64 time=5514.683 ms

--- 192.168.1.1 ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 5514.683/7014.865/8515.045 ms
root@OpenWrt:~# ping 192.168.1.10 -c 4
PING 192.168.1.10 (192.168.1.10): 56 data bytes
64 bytes from 192.168.1.10: seq=0 ttl=64 time=7629.289 ms
64 bytes from 192.168.1.10: seq=1 ttl=64 time=6629.080 ms
64 bytes from 192.168.1.10: seq=2 ttl=64 time=5628.935 ms
64 bytes from 192.168.1.10: seq=3 ttl=64 time=4628.800 ms

--- 192.168.1.10 ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 4628.800/6129.026/7629.289 ms
root@OpenWrt:~# ip n s
192.168.1.10 dev br-lan lladdr 00:07:XXXXXXXX REACHABLE
192.168.1.1 dev br-lan lladdr b8:69:XXXXXXXX STALE
192.168.1.128 dev br-lan lladdr f4:ce:XXXXXXXX REACHABLE
fe80::ba27:ebff:fec0:deba dev br-lan lladdr b8:27:XXXXXXXX STALE
fdc1:9f2e:acf3:5da8:ba27:ebff:fec0:deba dev br-lan lladdr b8:27:XXXXXXXX STALE
fe80::ba69:f4ff:fe72:da92 dev br-lan lladdr b8:69:XXXXXXXX router STALE
root@OpenWrt:~#
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

AgustinLorenzo, 2023-08-21:
Hi @Branyac 

Maybe, is related with that: https://forum.openwrt.org/t/ath11k-possible-firmware-bug-traffic-interruptions-when-a-client-disconnects-from-wlan/167400/37

Regards, Agustin

AgustinLorenzo, 2023-08-21:
Hi @Branyac 

Version removed, DHCP is broken I have to make a couple of changes.

Normally, when I publish a new "stable" version I post it in the Telegram group to avoid these problems.

Regards, Agustin

Branyac, 2023-08-23:
> Hi @Branyac
> 
> Maybe, is related with that: https://forum.openwrt.org/t/ath11k-possible-firmware-bug-traffic-interruptions-when-a-client-disconnects-from-wlan/167400/37
> 
> Please, join to Telegram group: https://t.me/ax3600_openwrt
> 
> Regards, Agustin

I tried the solutions but didn't work :(

Also, I tested with the latest release ( [Updated prebuilt images 2023-08-23-2106](https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-2023-08-23-2106) ) and it has the same bug.

Maybe it's a problem with the switch driver but I don't know how to probe it. Do you have any idea?

Thanks and regards.

AgustinLorenzo, 2023-09-20:
Hi @Branyac 

Fixed, in the latest version (BUILDING)

Regards, Agustin

closed 2023-09-20 by AgustinLorenzo