AgustinLorenzo/openwrt #5: Workaround of the build - ENGLISH

AgustinLorenzo, 2023-09-24:
Hi boys,

I open this issue to list the workarounds to make the version work correctly.

1st: DHCP does not deliver IP
  - Configure the IP on the interface: 192.168.1.100
  - Connect via SSH to the router: 192.168.1.1
  - Execute the following commands: uci set dhcp.lan.ignore=0 && uci commit && service dnsmasq restart

2nd: Problem with SSL, this problem is caused by ECM and where I have observed that the problem occurs the most is in connection with PPPoE, the solution is:
  - Enter LUCI
  - Go to Network -> Interfaces -> Devices, where it says pppoe-wan, click on edit and change the MTU to 1480, save and apply and the problem should disappear.

These are the registered errors at the moment.

Thanks for using my version :)

Regards, Agustin

antpir, 2023-12-30:
I have two Android TV on the IOT network which, even by applying the workaround in point 5, receive IP but do not browse, while if I connect them via wifi there is no problem.
`
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdde:add1:70b7::/48'
	option packet_steering '1'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'wan'
	option bridge_empty '1'

config interface 'lan'
	option device 'br-lan.10'
	option proto 'static'
	option ip6assign '60'
	option gateway '192.168.1.1'
	list dns '8.8.8.8'
	list dns '8.8.4.4'
	list ipaddr '192.168.1.8/24'

config interface 'IOT'
	option device 'br-lan.30'
	option proto 'static'
	option gateway '192.168.30.1'
	list ipaddr '192.168.30.2/24'
	list dns '8.8.8.8'
	list dns '8.8.4.4'
	option ip6assign '64'
	option ip4table 'local'

config interface 'GUEST'
	option device 'br-lan.20'
	option proto 'static'
	option ipaddr '192.168.20.2'
	option netmask '255.255.255.0'
	option gateway '192.168.20.1'
	list dns '8.8.8.8'
	list dns '8.8.4.4'

config bridge-vlan
	option device 'br-lan'
	option vlan '10'
	list ports 'lan3:t'
	list ports 'wan:t'

config bridge-vlan
	option device 'br-lan'
	option vlan '20'
	list ports 'lan3:t'
	list ports 'wan:t'

config bridge-vlan
	option device 'br-lan'
	option vlan '30'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3:t'
	list ports 'wan:t'

config bridge-vlan
	option device 'br-lan'
	option vlan '5'
	list ports 'lan3:t'
	list ports 'wan:t'

config device
	option name 'tun0'
	option acceptlocal '1'

config interface 'vpn'
	option proto 'none'
	option device 'tun0'
	list dns '8.8.8.8'
	list dns '8.8.4.4'
	option force_link '1'`

`
config dnsmasq
	option localise_queries '1'
	option rebind_protection '0'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option localservice '1'
	option ednspacket_max '1232'
	option confdir '/tmp/dnsmasq.d'
	option sequential_ip '1'
	option domainneeded '1'
	option authoritative '1'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

config dhcp 'lan'
	option interface 'lan'
	option limit '54'
	option leasetime '12h'
	option start '200'
	option force '1'
	list dhcp_option '3,192.168.1.1'
	list dhcp_option '6,8.8.8.8,8.8.4.4'
	option ignore '0'

config dhcp 'IOT'
	option interface 'IOT'
	option start '10'
	option limit '50'
	option leasetime '12h'
	list dhcp_option '3,192.168.30.1'
	list dhcp_option '6,8.8.8.8,8.8.4.4'
	option force '1'
	option ignore '0'

config dhcp 'GUEST'
	option interface 'GUEST'
	option start '100'
	option limit '150'
	option leasetime '12h'
	list dhcp_option '3,192.168.20.1'
	list dhcp_option '6,8.8.8.8,8.8.4.4'
	option force '1'`

AgustinLorenzo, 2024-04-27:
Attempt of solution for the punctual disconnections of WiFI devices:

Execute depending on the radios that our router has:

- AX3600: radio0 - radio1 - radio2
- AX9000: radio0 - radio1 - radio2 - radio3

```
uci set wireless.default_radio0.dtim_period='1'
uci set wireless.default_radio0.max_inactivity='86400'
uci set wireless.default_radio0.disassoc_low_ack='0'
uci set wireless.default_radio0.wnm_sleep_mode='1'

uci set wireless.default_radio1.dtim_period='1'
uci set wireless.default_radio1.max_inactivity='86400'
uci set wireless.default_radio1.disassoc_low_ack='0'
uci set wireless.default_radio1.wnm_sleep_mode='1'

uci set wireless.default_radio2.dtim_period='1'
uci set wireless.default_radio2.max_inactivity='86400'
uci set wireless.default_radio2.disassoc_low_ack='0'
uci set wireless.default_radio2.wnm_sleep_mode='1'

uci set wireless.default_radio3.dtim_period='1'
uci set wireless.default_radio3.max_inactivity='86400'
uci set wireless.default_radio3.disassoc_low_ack='0'
uci set wireless.default_radio3.wnm_sleep_mode='1'
```
Thanks to: https://forum.openwrt.org/t/qualcommax-nss-build/148529/2350

Regards, Agustin

AgustinLorenzo, 2024-10-20:
Good evening,

After the firmware change to NSS 12.2, this is no longer necessary.

Best regards, Agustin