AgustinLorenzo/openwrt #9: Steps to update to kernel 6.1 without losing configuration - ENGLISH

AgustinLorenzo, 2023-09-25:
Hi guys,

I leave you here, how to go from my latest version of kernel 5.15 to the latest kernel 6.1 (since without doing anything, it cannot be done due to the change from ipq807x to qualcommax), without losing the configuration and doing it through sysupgrade:

1. Connect via WinSCP or SSH (whichever is more convenient) to our router.
2. Edit the following file: /etc/openwrt_release
3. Modify the DISTRIB_TARGET line to the following value: `DISTRIB_TARGET='qualcommax/ipq807x'`
4. Now you can perform the update
    - 1st: Upload file to /tmp
    - 2nd: Connect via SSH if we are not previously connected and execute the following command
            - service wpad stop && sleep 5 && sysupgrade openwrt-qualcommax-ipq807x-xiaomi_ax3600-squashfs-sysupgrade.bin
    - 3rd: The router will restart automatically
5. After restarting, we may find that the DHCP service does not work. To solve it, do the following:
    - 1st: Connect to our network via WiFi/cable
    - 2nd: Give us an IP of our network
    - 3rd: Connect via SSH to our router and execute the following command: `uci set dhcp.lan.ignore=0 && uci commit && service dnsmasq restart`

With this everything would be done.

Regards, Agustin

strxno, 2023-09-26:
Can I upgrade from 2023-06-12-2358 to your latest build without wiping my config?

AgustinLorenzo, 2023-09-26:
Hi @strxno

Yes, just follow the steps

Thanks for using.

Regards, Agustín

GD2021, 2024-01-20:
I already have UBOOT, is it possible to directly flash your firmware on uboot?
[image]

yyleis, 2024-04-05:
> Hi guys,
>
> I leave you here, how to go from my latest version of kernel 5.15 to the latest kernel 6.1 (since without doing anything, it cannot be done due to the change from ipq807x to qualcommax), [...]

step 5 is important but it's diffcult to input it , because the wire lan only be connected about 3~5 seconds and then disconnected.
my router is qnap 301w, I quickly entered the command "fw_setenv current_entry 1" in the 3~5 seconds gap after boot, and then back to backup firmware.