AgustinLorenzo/openwrt #23: fail to use imagebuilder

thinkcyy, 2024-07-11:
No rule to make target 'redmi_ax6-initramfs-images'

AgustinLorenzo, 2024-07-21:
Hi @thinkcyy 

I'm sorry, but I don't understand what the problem is.

Regards, Agustin

thinkcyy, 2024-07-22:
I compiled imagebuilder from the latest commit 70555aa (at that time) for AX6, but failed to  generate fireware from this imagebuilder, with make image command reporting 
```
make[3]: *** No rule to make target 'redmi_ax6-initramfs-images', needed by '/.../openwrt-imagebuilder-qualcommax-ipq807x.Linux-x86_64/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/tmp/openwrt-qualcommax-ipq807x-redmi_ax6-squashfs-factory.ubi'.  Stop.
make[2]: *** [Makefile:208: build_image] Error 2
make[1]: *** [Makefile:146: _call_image] Error 2
```

It seems this issue happened since the following commits, cause the latest working imagebuilder I've test is commit ad3b782 (5th July).

* 0137fbd74b kernel: skip rebuilding kernel with ROOTFS_INITRAMFS_SEPARATE
* ec75f86612 image: respect TARGET_PER_DEVICE_ROOTFS for initramfs fit command
* 97fd059e7e image: respect TARGET_PER_DEVICE_ROOTFS for initramfs
* 00053299fa kernel: simplify SetInitramfs compression ALGO config setup
* 9b0185e0c0 kernel: make LINUX_DIR configurable for SetInitramfs functions
* 4209eb7505 kernel: allow different TARGET_DIR for initramfs

closed 2024-09-01 by AgustinLorenzo