AgustinLorenzo/openwrt #27: Stability of recent builds

Eloston, 2024-08-28:
I have been following your builds on the OpenWRT forums (https://forum.openwrt.org/t/qualcommax-nss-build/148529)

Normally when you make a release, I can see the corresponding update post on the above topic. However, the recent builds (2024-08-23 and now 2024-08-26), I have not seen the corresponding update posts and instead saw you have been reporting bugs from your builds.

Are any of the recent builds (2024-08-23 and 2024-08-26) ready for general use, or are they for testing only? Since GitHub marks those Releases as the latest on your repository, I am not sure if I should be upgrading or not. If they are not ready for general use, could you please mark them as Pre-release on GitHub? It is an option you can set on individual releases.

Much appreciated, and thank you for your consistent effort and quality on these builds. I have been daily driving them on my Dynalink, and they have been incredibly stable and performant compared to vanilla OpenWRT.

AgustinLorenzo, 2024-09-01:
Hi @Eloston,

Thanks, for opening an issue :)

The version of 08/23/2024 is stable and is published in the OpenWRT forum, it only has a bug that causes a “kernel panic” in some IPTV scenarios (which at the moment only occur with Movistar ISP in Spain).

The version of 08/26/2024 is also stable only includes these 2 different commits (https://github.com/AgustinLorenzo/openwrt/commit/c965b5006f7b715ab6a42aa5511b0499a083a122) and (https://github.com/AgustinLorenzo/openwrt/commit/682b7840b7075d2e288d54c1996f47416079eb4c), the first commit is only to remove NSS modules that we do not use and the second commit adds the modules to get the crashlogs that I mention in this post (https://forum.openwrt.org/t/qualcommax-nss-build/148529/3674?u=agustinlorenzo).

Regards, Agustin

AgustinLorenzo, 2024-09-01:
Hi all,

The only thing I would ask is that if you are on the version of the day 08/26/2024 and your router appears here (https://github.com/openwrt/openwrt/pull/15688) as untested and you can reproduce what qosmio indicates in that thread, comment and indicate what the device is and that it is working properly, this will help qosmio to bring this to the main branch of OpenWRT.

On ax3600 is working fine :)

Regards, Agustin

Eloston, 2024-09-02:
Thanks for the details! Feel free to close the issue.

AgustinLorenzo, 2024-09-08:
Hi @Eloston 

New version released and published in the forums, I close the issue.

Regards, Agustin

closed 2024-09-08 by AgustinLorenzo