AgustinLorenzo/openwrt #30: No es posible instalar software mediante opkg

hav4k, 2024-11-13:
Hola.

Según parece, han cambiado el gestor de paquetes a apk. ¿Alguna solución para la última build de la semana pasada? Gracias.

Un saludo.

gnacho, 2024-11-13:
No sé si está relacionado, pero a mi no me deja instalar NADA.

[image]

He probado la imagen anterior y tampoco ha sido posible.

AgustinLorenzo, 2024-11-17:
Buenas,

Efectivamente con el cambio de OPKG a APK se rompieron todos los respo de OpenWRT de SNAPSHOT, acabo de lanzar build ya basada en APK y ya vuelve a funcionar.

Se ha comentado esto mismo en telegram: https://t.me/ax3600_openwrt

Cierro.

Un saludo, Agustin

closed 2024-11-17 by AgustinLorenzo