AgustinLorenzo/openwrt #31: Add MAP and DS-lite to Buffalo WXR-5950AX12 builds

lvctr, 2024-11-17:
Buffalo WXR-5950AX12 is a router sold in Japan for Japanese market.

In Japan, for 10G we use MAP-E or DS-lite to connect to the internet.

Since the builds don't include MAP and DS-lite packages, they are not usable by the intended user base for these builds.

As of the moment we have been building by ourselves, but having it added to the 5950AX12 config and having prebuilts with these packages would be extremely helpful for all of us in Japan.

Thank you!

AgustinLorenzo, 2024-11-18:
Hi @lvctr,

I think you are in an old version because the packages for MAP-T are added since a few versions ago, you can try in the last one in case it doesn't work tell me what they are and I'll check them.

Regards, Agustin

lvctr, 2024-11-18:
Hi, thank you for the response.

Yes, MAP-T packages for NSS are included, but MAP-T is not MAP-E.

MAP-E is included in `map` package which is not included. In the `.full_config` file as well it can be seen that `CONFIG_PACKAGE_map is not set`. `ds-lite` package is also not included. These cannot be installed after the fact because they require specific kmod dependencies which need to be included at build time (the official OpenWRT releases do this).

I don't mind making a pull request myself.

lvctr, 2024-11-18:
Hi, thank you for the response.

Yes, MAP-T packages for NSS are included, but MAP-T is not MAP-E.

MAP-E is included in map package which, unless I'm misinterpreting something, is not included. In the .full_config file as well it can be seen that CONFIG_PACKAGE_map is not set. ds-lite package is also not included.

I don't mind making a pull request myself!

closed 2024-11-23 by AgustinLorenzo