AgustinLorenzo/openwrt #33: NSS wifi 2024-11-23-2203 : IPV6 internet connection not established.

mrlamud, 2024-11-24:
Sysupgraded from 2024-11-17-1854 ( which is the best so far) to 2024-11-23-2203. 
IPV 4 & 6 DHCP is working normal but internet connection only works for IPV4  address.

So I downgraded to 2024-11-17-1854 for now.

Thanks for your dedication to the community.

AgustinLorenzo, 2024-12-07:
Hi @mrlamud,

Can you retest in the latest?

Regards, Agustin

mrlamud, 2024-12-07:
I had tested the lates, the result is still the same.
The router can obtain ipv6 from  wan but dhcp does not  assign ipv6 to all clients.

BTW. - I just flash sysupgrade. Neither keep all configs & settings nor wipe all and reconfig but yeild the same result - no ipv6 assigned.

mrlamud, 2024-12-08:
More information.

I decided to return to the ax3600 stock firmware and then flash 2024-12-07 (NSS wifi) but the result remains the same, can not connect ipv6 internet sites.

IPV6 wan obtained, router assigns ipv6 to client as normal.

AgustinLorenzo, 2024-12-08:
Hi @mrlamud,

I found the reason for the IPv6 breakage, it is related to the mwan3 package that I recently added, to avoid this kind of behavior I will release a new build with this service disabled.

Give me a couple of hours and it's solved.

Regards, Agustin

mrlamud, 2024-12-08:
Thank you very much, Agustin.

AgustinLorenzo, 2024-12-08:
Hi @mrlamud 

Building, link: https://github.com/AgustinLorenzo/openwrt/actions/runs/12226296486/job/34101332489

I  have a TG channel, join to that: https://t.me/ax3600_openwrt

Regards, Agustin

mrlamud, 2024-12-09:
ipq807x-nsswifi-2024-12-09-0128 works .
Thanks for TG channel, I'd joined.

closed 2024-12-09 by AgustinLorenzo