AgustinLorenzo/openwrt #40: ath11k Driver "binary attribute size out of range" Error Causes STA Authentication Failures

jason5545, 2025-02-21:
Description:
I'm experiencing repeated authentication failures when using the ath11k driver in STA mode on OpenWrt version 256de27. When attempting to connect to a wireless network (e.g., SSID "GL-MT6000_5G" with BSSID 94:83:::**:3A), the connection fails with the following error:

nl80211: kernel reports: binary attribute size out of range

This error prevents the 4-way handshake from completing and stops the client from successfully associating with the access point—even though the password is correct.

Steps to Reproduce:

1. Install OpenWrt version 256de27 on a device equipped with an ath11k-based Wi‑Fi chipset.

2. Configure the device to operate in STA mode and set it up to connect to the wireless network "GL-MT6000_5G" (BSSID 94:83:::**:3A) with the correct password.

3. Observe the logs during the authentication process.

4. Notice repeated errors stating "binary attribute size out of range" along with messages such as "Authentication request to the driver failed," and the client being temporarily disabled (added to an ignore list).

Expected Behavior:
The wireless client should successfully complete the 4-way handshake and associate with the access point.

Actual Behavior:
The ath11k driver returns a "binary attribute size out of range" error during the authentication process, causing connection attempts to fail repeatedly.

Environment Information:

OpenWrt Version: 256de27

Hardware Model: Linksys MX4200v1

Additional Notes:

This issue appears only in STA mode; AP mode operates normally.

I have not modified the wpa_supplicant version as it does not seem to be the source of the problem.

Increasing the debug level for logging in both wpa_supplicant and the kernel may provide further insights.

Any guidance or patches to resolve this incompatibility would be greatly appreciated.

Thank you for your assistance!

---

gittyaf, 2025-02-22:
Issue is known to have started around September:

https://github.com/qosmio/openwrt-ipq/issues/21

I confirmed this working again after reflashing an early August build from this repo on MX4200v1.

AgustinLorenzo, 2025-03-15:
Hi,

This is corrected in the latest version.

Regards, Agustin

closed 2025-03-15 by AgustinLorenzo