AgustinLorenzo/openwrt #43: Need kmods for batman

thecuriousgeek, 2025-03-20:
Any way you can publish the kmods like [arix](https://github.com/arix00/openwrt-mx4300. I need the batman kmod to use batman-adv with your build
Need the kmods compatible with the overall kernel and other modules, so cant use his

BTW, been using your build (from 2/17/25) for over a month now with absolutely no issues, thank you very much

AgustinLorenzo, 2025-04-03:
Hi @thecuriousgeek,

I will take a look at how I can add the kmods to the build without generating too much work.

Regards, Agustin

closed 2025-04-03 by AgustinLorenzo

Snuupy, 2025-04-18:
@AgustinLorenzo perhaps you missed my comment here https://forum.openwrt.org/t/qualcommax-nss-build/148529/4400?u=snuupy so I will repost:

Could you please add this line https://github.com/arix00/openwrt-mx4300/blob/79024d0fc08b5315765a280d1670ecbc77b890e8/include/feeds.mk#L41 to the build so that we can get kmods so we don't have to rebuild the firmware ourselves? Thank you ❤