AgustinLorenzo/openwrt #44: coreutils 9.6 fails to build

deepakvinod, 2025-03-20:
Hey,

With the latest tag, coreutils got updated to 9.6 and fails to build.

```
 make[2] -C feeds/packages/utils/coreutils compile
configure: WARNING: unrecognized options: --with-gmp
configure: WARNING: libsmack support disabled by user
configure: WARNING: result yes guessed because of cross compilation
configure: WARNING: 'su' is not an optionally-installable program
configure: WARNING: unrecognized options: --with-gmp
lib/randint.c: In function 'randint_all_new':
lib/randint.c:87:43: error: 'nullptr' undeclared (first use in this function); did you mean 'nullptr_t'?
   87 |   return (source ? randint_new (source) : nullptr);
      |                                           ^~~~~~~
      |                                           nullptr_t
lib/randint.c:87:43: note: each undeclared identifier is reported only once for each function it appears in
make[5]: *** [Makefile:17161: lib/libcoreutils_a-randint.o] Error 1
make[5]: *** Waiting for unfinished jobs....
lib/randperm.c: In function 'sparse_new':
lib/randperm.c:101:38: error: 'nullptr' undeclared (first use in this function); did you mean 'nullptr_t'?
  101 |   return hash_initialize (size_hint, nullptr, sparse_hash_, sparse_cmp_, free);
      |                                      ^~~~~~~
      |                                      nullptr_t
lib/randperm.c:101:38: note: each undeclared identifier is reported only once for each function it appears in
lib/randperm.c: In function 'randperm_new':
lib/randperm.c:156:11: error: 'nullptr' undeclared (first use in this function); did you mean 'nullptr_t'?
  156 |       v = nullptr;
      |           ^~~~~~~
      |           nullptr_t
make[5]: *** [Makefile:17175: lib/libcoreutils_a-randperm.o] Error 1
lib/root-dev-ino.c: In function 'get_root_dev_ino':
lib/root-dev-ino.c:33:12: error: 'nullptr' undeclared (first use in this function); did you mean 'nullptr_t'?
   33 |     return nullptr;
      |            ^~~~~~~
      |            nullptr_t
lib/root-dev-ino.c:33:12: note: each undeclared identifier is reported only once for each function it appears in
make[5]: *** [Makefile:17469: lib/libcoreutils_a-root-dev-ino.o] Error 1
lib/randread.c: In function 'randread_new':
lib/randread.c:176:24: error: 'nullptr' undeclared (first use in this function); did you mean 'nullptr_t'?
  176 |     return simple_new (nullptr, nullptr);
      |                        ^~~~~~~
      |                        nullptr_t
lib/randread.c:176:24: note: each undeclared identifier is reported only once for each function it appears in
lib/randread.c: In function 'readisaac':
lib/randread.c:51:58: error: implicit declaration of function 'alignof' [-Wimplicit-function-declaration]
   51 | # define POINTER_IS_ALIGNED(ptr, type) ((size_t) (ptr) % alignof (type) == 0)
      |                                                          ^~~~~~~
lib/randread.c:284:11: note: in expansion of macro 'POINTER_IS_ALIGNED'
  284 |       if (POINTER_IS_ALIGNED (p, isaac_word))
      |           ^~~~~~~~~~~~~~~~~~
lib/randread.c:284:34: error: expected expression before 'isaac_word'
  284 |       if (POINTER_IS_ALIGNED (p, isaac_word))
      |                                  ^~~~~~~~~~
lib/randread.c:51:67: note: in definition of macro 'POINTER_IS_ALIGNED'
   51 | # define POINTER_IS_ALIGNED(ptr, type) ((size_t) (ptr) % alignof (type) == 0)
      |                                                                   ^~~~
make[5]: *** [Makefile:17189: lib/libcoreutils_a-randread.o] Error 1
make[4]: *** [Makefile:23082: all-recursive] Error 1
make[3]: *** [Makefile:9124: all] Error 2
make[2]: *** [Makefile:154: /home/dev/data/OpenWRT/DynalinkDLWRX36/openwrt-alorenzo/build_dir/target-aarch64_cortex-a53_musl/coreutils-9.6/.built] Error 2
    ERROR: package/feeds/packages/coreutils failed to build.
```

I'm building on a debian distrobox on linux. Tried distclean which did not help.

AgustinLorenzo, 2025-04-03:
Hi @deepakvinod,

I have re-synchronized the repository with OpenWRT should be fixed, I'm compiling.

Regards, Agustin

closed 2025-04-03 by AgustinLorenzo

deepakvinod, 2025-04-03:
Thank you! It builds fine now. 
Best,
Deepak