AgustinLorenzo/openwrt #51: Arcadyan AW1000 QMI Cellular Speed Limitation

alexhee79, 2025-05-10:
Before this is using other OpenWRT fork (OpenWRT, KWRT & ImmortalWRT) in QMI cellular mode for my 5G SIM, able to achieve average 500Mbps for download & around 90Mbps for download.

However, this NSS build version only able to reach 250Mbps  for download & around 90Mbps for upload. It seems the download speed  has been cut half but unsure whether this is my settings issue or else?

Anyone with AW1000 is having the same issue as me?

Modem: Arcadyna AW1000
ISP: Yes 5G
Firmware: Updated prebuilt images (NSS-WiFi) 2025-05-06-0433

AgustinLorenzo, 2025-05-12:
Hi @alexhee79,

Is it a build compiled by you or is it in the builds that I compile, because in those builds the wwan acceleration is not added,

Regards, Agustin

alexhee79, 2025-05-12:
Hi, I'm unaware that NSS is not built on your firmware. I'm using your firmware not own compiled. Then it's a bit puzzle that I'm using OpenWrt official build with better speed but not yours.

If you have free time care to take a look? Thanks.

alexhee79, 2025-05-12:
Reopen for comment

AgustinLorenzo, 2025-05-15:
I see how I can manage it :)

closed 2025-05-15 by AgustinLorenzo