AgustinLorenzo/openwrt #52: Build for tag ipq807x-nsswifi-2025-05-12-1225 breaks WAN DHCP client

Eloston, 2025-05-13:
On my Dynalink DL-WRX36 with mwan3 disabled, my WAN interface (using DHCP and DHCPv6) cannot get an address via DHCP from my ISP when using the build [`ipq807x-nsswifi-2025-05-12-1225`](https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-nsswifi-2025-05-12-1225) (the current latest build). However, downgrading to the previous version [`ipq807x-nsswifi-2025-05-06-0433`](https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-nsswifi-2025-05-06-0433) restores functionality.

Eloston, 2025-05-14:
I didn't try with official snapshot, but IMO it doesn't really narrow down the problem that much (is it caused by NSS changes, or a specific version of a package?)

I use AT&T and both IPv4 and IPv6 worked fine in all previous builds from this repo (I use non-mesh only). This is the first build where it couldn't get any addresses via DHCP no matter what I tried.

MiguVT, 2025-05-14:
Not sure what it can be but the history of AT&T isnt good about IPV6, they
usually dont follow IPV6 standard and cant correctly comunicate with RA.
Sometimes there is a known walkthrough, if not you should error and try.
Personally I dont know much about the current situation of AT&T.

Eloston, 2025-05-14:
yeah but the main problem is IPv4 and IPv6 are BOTH broken, so it seems likely a networking or DHCP client issue caused by a bad package (IMO more likely) or a broken NSS change.

If it was just IPv6 then yeah, but I kept rebooting the router a few times and restarting the interfaces, and I couldn't even get IPv4 addressing. But it works on the old firmware version and also my personal Linux laptop.

MiguVT, 2025-05-15:
I dont have any problem with IPV4, weird (Using a build from a week ago)

AgustinLorenzo, 2025-05-15:
Hi,

In my setup, I have not encountered any problems:
- AX3600: VLAN 20 (DIGI SPAIN) + PPPoE (I get IPv4) and in turn via DHCPv6 (I get IPv6) working correctly.
- AX3600: bridge mode (Vodafone Spain): only IPv4 via DHCP, working correctly.

Regards, Agustin

MiguVT, 2025-05-15:
> Hi,
> 
> In my setup, I have not encountered any problems:
> 
> * AX3600: VLAN 20 (DIGI SPAIN) + PPPoE (I get IPv4) and in turn via DHCPv6 (I get IPv6) working correctly.
> * AX3600: bridge mode (Vodafone Spain): only IPv4 via DHCP, working correctly.
> 
> Regards, Agustin

Oh, IPv6 is working correctly for you?? whats your settings? I have DIGI SPAIN and they arent giving me IPv6 correctly (It dont works after reboot or sometimes, works weird), it looks like its set and after some seconds it dissapear.

AgustinLorenzo, 2025-05-15:
> Oh, IPv6 is working correctly for you?? whats your settings? I have DIGI SPAIN and they arent giving me IPv6 correctly (It dont works after reboot or sometimes, works weird), it looks like its set [...]

Buenas,

Entiendo que eres español, el IPv6 de DIGI es tremendamente sensible, solo funcionamiento correctamente si se aprovisiona a la vez que arranca el PPPoE, si reinicias después o cualquier cosa se pierde.

La configuración que tengo es la siguiente:

- WAN: protocol (pppoe), device(wan.20), y desde advanced settings, pones "obtain ipv6 address" en disabled
- WAN6: protocol (dhcpv6 client), device (pppoe-wan)

Guardas, aplicas y reinicias.

Con esto aparte de dejar el IPv6 correctamente configurado y gestionable, evitas la interfaz gris que genera automáticamente el PPPoE.

PD: Tenemos grupo de Telegram: https://t.me/ax3600_openwrt

Un saludo, Agustin

MiguVT, 2025-05-15:
> > > Hi,
> > > In my setup, I have not encountered any problems:
> > > 
> > > * AX3600: VLAN 20 (DIGI SPAIN) + PPPoE (I get IPv4) and in turn via DHCPv6 (I get IPv6) working correctly.
> > > * AX3600: bridge mode (Vodafone Spain): only IPv4 via DHCP, working correctly.
> > > 
> > > Regards, Agustin
> > 
> > 
> > Oh, IPv6 is working correctly for you?? whats your settings? I have DIGI SPAIN and they arent giving me IPv6 correctly (It dont works after reboot or sometimes, works weird), it looks like its set and after some seconds it dissapear.
> 
> Buenas,
> 
> Entiendo que eres español, el IPv6 de DIGI es tremendamente sensible, solo funciona correctamente si se aprovisiona a la vez que arranca el PPPoE, si reinicias después la interfaz o cualquier cosa se pierde.
> 
> La configuración que tengo es la siguiente:
> 
> * WAN: protocol (pppoe), device(wan.20), y desde advanced settings, pones "obtain ipv6 address" en disabled
> * WAN6: protocol (dhcpv6 client), device (pppoe-wan)
> 
> Guardas, aplicas y reinicias.
> 
> Con esto aparte de dejar el IPv6 correctamente configurado y gestionable, evitas la interfaz gris que genera automáticamente el PPPoE.
> 
> PD: Tenemos grupo de Telegram: https://t.me/ax3600_openwrt
> 
> Un saludo, Agustin

Oh entiendo, muchas gracias cuando pueda lo aplicare, me imagino que mi unico problema era que tenia obtain ipv6 address en manual. Gracias.

AgustinLorenzo, 2025-05-17:
Hi @Eloston,

I can confirm the bug I am working on fixing it.

Is related to this commit: https://github.com/AgustinLorenzo/openwrt/commit/cc50cac8a0fd0e12fc9e2a7fa7d14f99a2a5ffae

Regards, Agustin

AgustinLorenzo, 2025-05-19:
Hi @Eloston,

Is fixed in the latest released version.

Regards, Agustin

closed 2025-05-19 by AgustinLorenzo

Eloston, 2025-05-19:
Thanks for the quick fix!