AgustinLorenzo/openwrt #53: no phy3 wifi (Qualcomm Atheros QCN6024/9024/9074 802.11ac/ax/n) on latest build (b8ada65) with AX9000

sudousercom, 2025-05-19:
Hi! build - https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-nsswifi-mesh-2025-05-19-0315
phy3 just showing as generic radio and wont start

and another question, i dont have strong devops skills, tried to build for ax900 only from your repo, but with no luck, no wifi drivers was in firware, but maybe you can add ax9000 specific packages, the needed for correct fan work,

kernel modules needed:
`kmod-hwmon-core kmod-thermal kmod-regmap-i2c kmod-i2c-core kmod-regmap-core kmod-hwmon-emc2305`

packages:
https://github.com/devrus91/openwrt-alpine-fan-control

Thanks!

AgustinLorenzo, 2025-05-20:
Hi @sudousercom,

Did you check the openwrt wiki?

Link: https://openwrt.org/toh/xiaomi/ax9000#potential_issueslimitations

Regards, Agustin

sudousercom, 2025-05-21:
yes, i have working board-2.bin file, but adatapter is unrecognized by system builded with .full-config from repo

sudousercom, 2025-05-21:
looks like .config has drivers disabled by default, .full_config in repo, here dmesg from my buld

```[   12.378211] ath10k 6.14 driver, optimized for CT firmware, probing pci device: 0x50.
[   12.378650] ath10k_pci 0001:01:00.0: enabling device (0000 -> 0002)
[   12.385520] ath10k_pci 0001:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   13.067400] ath10k_pci 0001:01:00.0: Failed to find firmware-N.bin (N between 2 and 6) from ath10k/QCA9887/hw1.0: -12
[   13.067454] ath10k_pci 0001:01:00.0: could not fetch firmware files (-12)
[   13.077044] ath10k_pci 0001:01:00.0: could not probe fw (-12)
[   13.096742] ath11k c000000.wifi: ipq8074 hw2.0
[   13.096775] ath11k c000000.wifi: FW memory mode: 0
[   13.128565] remoteproc remoteproc0: powering up cd00000.q6v5_wcss
[   13.132665] remoteproc remoteproc0: Booting fw image IPQ8074/q6_fw.mdt, size 668
[   14.086848] remoteproc remoteproc0: remote processor cd00000.q6v5_wcss is now up
[   14.131718] ath11k c000000.wifi: qmi fail to get qcom,m3-dump-addr, ignore m3 dump mem req
[   14.138176] ath11k c000000.wifi: chip_id 0x0 chip_family 0x0 board_id 0xff soc_id 0xffffffff
[   14.138880] ath11k c000000.wifi: fw_version 0x290b84a5 fw_build_timestamp 2024-09-23 11:32 fw_build_id WLAN.HK.2.9.0.1-02146-QCAHKSWPL_SILICONZ-1
[   14.276518] remoteproc remoteproc0: stopped remote processor cd00000.q6v5_wcss
[   14.276561] remoteproc remoteproc0: powering up cd00000.q6v5_wcss
[   14.282725] remoteproc remoteproc0: Booting fw image IPQ8074/q6_fw.mdt, size 668
[   14.632383] remoteproc remoteproc0: remote processor cd00000.q6v5_wcss is now up
[   14.634092] ath11k_pci 0000:01:00.0: BAR 0: assigned [mem 0x20400000-0x205fffff 64bit]
[   14.638933] ath11k_pci 0000:01:00.0: enabling device (0000 -> 0002)
[   14.647060] ath11k_pci 0000:01:00.0: MSI vectors: 16
[   14.652764] ath11k_pci 0000:01:00.0: qcn9074 hw1.0
[   14.657949] ath11k_pci 0000:01:00.0: FW memory mode: 2
[   14.676343] ath11k c000000.wifi: qmi fail to get qcom,m3-dump-addr, ignore m3 dump mem req
[   14.683802] ath11k c000000.wifi: chip_id 0x0 chip_family 0x0 board_id 0xff soc_id 0xffffffff
[   14.683853] ath11k c000000.wifi: fw_version 0x290b84a5 fw_build_timestamp 2024-09-23 11:32 fw_build_id WLAN.HK.2.9.0.1-02146-QCAHKSWPL_SILICONZ-1
[   14.849777] mhi mhi0: Requested to power ON
[   14.849836] mhi mhi0: Power on setup success
[   14.881468] ath11k c000000.wifi: htt event 48 not handled
[   14.884498] ath11k c000000.wifi: nss init soc nss if_num 203 userpd_id 0
[   15.407908] mhi mhi0: Wait for device to enter SBL or Mission mode
[   15.458884] l2tp_ppp: PPPoL2TP kernel driver, V2.0
[   15.461141] usbcore: registered new interface driver option
[   15.462671] usbserial: USB Serial support registered for GSM modem (1-port)
[   15.472825] kmodloader: done loading kernel modules from /etc/modules.d/*
[   15.479314] ath11k_pci 0000:01:00.0: chip_id 0x0 chip_family 0x0 board_id 0xff soc_id 0xffffffff
[   15.481984] ath11k_pci 0000:01:00.0: fw_version 0x290b8862 fw_build_timestamp 2024-09-23 10:51 fw_build_id
[   15.530975] ath11k_pci 0000:01:00.0: failed to fetch board data for bus=pci,qmi-chip-id=0,qmi-board-id=255 from ath11k/QCN9074/hw1.0/board-2.bin
[   15.531031] ath11k_pci 0000:01:00.0: failed to fetch board data for bus=pci,qmi-chip-id=0,qmi-board-id=255 from ath11k/QCN9074/hw1.0/board-2.bin
[   15.543210] ath11k_pci 0000:01:00.0: failed to fetch board.bin from QCN9074/hw1.0
[   15.555977] ath11k_pci 0000:01:00.0: qmi failed to fetch board file: -12
[   15.563474] ath11k_pci 0000:01:00.0: failed to load board data file: -12
[   17.627910] ECM init```

and ath10-11 from my .config

```CONFIG_DEFAULT_ath11k-firmware-ipq8074=y
CONFIG_DEFAULT_ath11k-firmware-qcn9074=y
CONFIG_DEFAULT_kmod-ath11k-ahb=y
CONFIG_DEFAULT_kmod-ath11k-pci=y
# CONFIG_PACKAGE_ath11k-firmware-ipq5018 is not set
# CONFIG_PACKAGE_ath11k-firmware-ipq6018 is not set
CONFIG_PACKAGE_ath11k-firmware-ipq8074=y
# CONFIG_PACKAGE_ath11k-firmware-qca2066 is not set
# CONFIG_PACKAGE_ath11k-firmware-qca6390 is not set
CONFIG_PACKAGE_ath11k-firmware-qcn9074=y
# CONFIG_PACKAGE_ath11k-firmware-wcn6750 is not set
# CONFIG_PACKAGE_ath11k-firmware-wcn6855 is not set
CONFIG_PACKAGE_kmod-ath11k=y
CONFIG_PACKAGE_kmod-ath11k-ahb=y
CONFIG_PACKAGE_kmod-ath11k-pci=y
CONFIG_DEFAULT_ath10k-firmware-qca9887=y
CONFIG_DEFAULT_kmod-ath10k=y
# ath10k Board-Specific Overrides
# end of ath10k Board-Specific Overrides
# CONFIG_PACKAGE_ath10k-board-qca4019 is not set
# CONFIG_PACKAGE_ath10k-board-qca9377 is not set
CONFIG_PACKAGE_ath10k-board-qca9887=y
[... 7 lines cut ...]
# CONFIG_PACKAGE_ath10k-firmware-qca4019-ct-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca6174 is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9377 is not set
CONFIG_PACKAGE_ath10k-firmware-qca9887=m
CONFIG_PACKAGE_ath10k-firmware-qca9887-ct=m
CONFIG_PACKAGE_ath10k-firmware-qca9887-ct-full-htt=m
# CONFIG_PACKAGE_ath10k-firmware-qca9888 is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9888-ct is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9888-ct-full-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9888-ct-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca988x is not set
# CONFIG_PACKAGE_ath10k-firmware-qca988x-ct is not set
# CONFIG_PACKAGE_ath10k-firmware-qca988x-ct-full-htt is not set
CONFIG_PACKAGE_ath10k-firmware-qca9984=m
# CONFIG_PACKAGE_ath10k-firmware-qca9984-ct is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9984-ct-full-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca9984-ct-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca99x0 is not set
# CONFIG_PACKAGE_ath10k-firmware-qca99x0-ct is not set
# CONFIG_PACKAGE_ath10k-firmware-qca99x0-ct-full-htt is not set
# CONFIG_PACKAGE_ath10k-firmware-qca99x0-ct-htt is not set
CONFIG_PACKAGE_kmod-ath10k=y
CONFIG_PACKAGE_kmod-ath10k-ct=y
# CONFIG_PACKAGE_kmod-ath10k-ct-smallbuffers is not set
# CONFIG_PACKAGE_kmod-ath10k-smallbuffers is not set```

ascendbeing, 2025-06-02:
@sudousercom Are the =m packages installed? by default you would need to install them, they wouldn't have been built into your image. only if they're =Y will they be installed into the image without manually installing them
the packages should be available in bin dir if you find you don't have em installed.

I don't know about this specific device but if you need those you might want them to be =y. and that may resolve it.

sudousercom, 2025-06-03:
tried with =y packages, all radios except QCN9074 work, firmware files for QCN are present, replaced board-2.bin with a working one, but without success

dmesg:
```[   15.816914] ath11k_pci 0000:01:00.0: BAR 0: assigned [mem 0x20400000-0x205fffff 64bit]
[   15.821740] ath11k_pci 0000:01:00.0: enabling device (0000 -> 0002)
[   15.829888] ath11k_pci 0000:01:00.0: MSI vectors: 16
[   15.835560] ath11k_pci 0000:01:00.0: qcn9074 hw1.0
[   15.840755] ath11k_pci 0000:01:00.0: FW memory mode: 2
[   15.859202] ath11k c000000.wifi: qmi fail to get qcom,m3-dump-addr, ignore m3 dump mem req
[   15.866608] ath11k c000000.wifi: chip_id 0x0 chip_family 0x0 board_id 0xff soc_id 0xffffffff
[   15.866658] ath11k c000000.wifi: fw_version 0x290b84a5 fw_build_timestamp 2024-09-23 11:32 fw_build_id WLAN.HK.2.9.0.1-02146-QCAHKSWPL_SILICONZ-1
[   16.634809] kmodloader: done loading kernel modules from /etc/modules.d/*
[   16.644402] ath11k_pci 0000:01:00.0: chip_id 0x0 chip_family 0x0 board_id 0xff soc_id 0xffffffff
[   16.647598] ath11k_pci 0000:01:00.0: fw_version 0x290b8862 fw_build_timestamp 2024-09-23 10:51 fw_build_id
[   16.691585] ath11k_pci 0000:01:00.0: failed to fetch board data for bus=pci,qmi-chip-id=0,qmi-board-id=255 from ath11k/QCN9074/hw1.0/board-2.bin
[   16.691641] ath11k_pci 0000:01:00.0: failed to fetch board data for bus=pci,qmi-chip-id=0,qmi-board-id=255 from ath11k/QCN9074/hw1.0/board-2.bin
[   16.703677] ath11k_pci 0000:01:00.0: failed to fetch board.bin from QCN9074/hw1.0
[   16.716759] ath11k_pci 0000:01:00.0: qmi failed to fetch board file: -12
[   16.723978] ath11k_pci 0000:01:00.0: failed to load board data file: -12
[   18.760828] ECM init```

AgustinLorenzo, 2025-06-05:
I understand that you only want to compile for AX9000 in that case I advise you to uncheck all other devices and disable the multidevices option and check AX9000 from “make menuconfig” to avoid strange behaviors.

closed 2025-06-06 by AgustinLorenzo