AgustinLorenzo/openwrt #56: Built-in updates

DustyTT, 2025-05-29:
Since this looks like a very custom build of OpenWRT, i'm guessing i should leave the updates available _System_ -> _Software_ alone and just stick with sys-updates available from his repo?

ascendbeing, 2025-06-02:
Probably. You could check /etc/opkg and see where it's sourcing the files from. if it looks like it's sourcing them from owrt.org and not a repo associated with this project, then yeah probably.
this is probably the most accessible nss fork cause of packaging basically just about everything you might typically need. if you're feeling adventurous there is a more up to date nss 6.12 based fork but you would have to compile it unless someone's already picked it up.

AgustinLorenzo, 2025-06-05:
Hi @DustyTT and @ascendbeing 

The first builds (regular and mesh) with kernel 6.12 are already released.

Regular: https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-nsswifi-2025-06-04-0300
Mesh: https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-nsswifi-mesh-2025-06-04-1609

Regards, Agustin

closed 2025-06-05 by AgustinLorenzo