AgustinLorenzo/openwrt #61: DL-WRX36 radios link-down on 2025-07-02-0342 build

aedancullen, 2025-07-03:
The latest NSS-WiFi build 2025-07-02-0342 introduces an issue on Dynalink DL-WRX36 where both radios (I believe) will instantly go to "link down" state upon a certain station associating (for me it's a Pixel 7a Android phone on 5GHz.) This does not happen with all stations. 

Surely there is some distinctive aspect of the station that is causing something to blow up, but unfortunately I can't debug yet as I'm traveling currently. Maybe someone else can reproduce.

mrlamud, 2025-07-03:
I found that sysupgrade without keep old config &  restore old config from backup archive solve this problem for me.

MiguVT, 2025-07-03:
I tried and no results :c stills the same

mrlamud, 2025-07-04:
> I tried and no results :c stills the same

What I did was  reverting to previous version (using firmware upgrade (or cli sysupgrade in the terminal) , backup config then do the firmware upgrade (or sysupgrade in the terminal) this latest without keep the config. After reboot then restore the previous config.

MiguVT, 2025-07-04:
Still not working , i tried fresh install and not worked :c

senorkrabs, 2025-07-04:
Seeing similar behavior on mx4300. Sending a large amount of traffic (iperf3) caused the radios to restart.

MiguVT, 2025-07-05:
> > I went to the month ago release and everything working fine (except my ax3600 is like a christmas tree, visual bug, or maybe a feature 😆)
> 
> Hi,
> 
> There seems to be something wrong with the LEDs with kernel 6.12, you should report it in the upstream as it also happens in the vanilla versions.
> 
> Regards, Agustin

At this time an option to do it with the leds could be a good feature, just funny how a bug can be an idea for a feature, i will report it when able. tysm <3

AgustinLorenzo, 2025-07-05:
Hi,

New build: https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-nsswifi-2025-07-05-1733

But I think there is something broken in the OpenWRT part, LUCI is not loading correctly.

Regards

javykhan, 2025-07-05:
> Hi,
> 
> New build: https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-nsswifi-2025-07-05-1733
> 
> But I think there is something broken in the OpenWRT part, LUCI is not loading correctly.
> 
> Regards

Yes luci overview page is still not loading.

MiguVT, 2025-07-05:
maybe mark all this releases as pre-release is a good thing to do for preventing people to install a unstable version

AgustinLorenzo, 2025-07-07:
Hi @mrlamud @aedancullen @senorkrabs @MiguVT 

Can you try this build and let me know how it goes?

Link: https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-nsswifi-2025-07-07-0218

The only thing I have seen is what I am commenting here: https://github.com/openwrt/openwrt/commit/bb279e1a69b17d68d4cab299c4bb290a948b9930#comments

Regards, Agustin

PhoenixG36, 2025-07-07:
Not sure if my last log was helpful.
So I did a new system log.

[log2.txt](https://github.com/user-attachments/files/21102015/log2.txt)

javykhan, 2025-07-08:
Whats changed in new release?

AgustinLorenzo, 2025-07-09:
Hi @mrlamud @aedancullen @senorkrabs @MiguVT 

The latest version is now stable and everything is stable again.

The changes were to synchronize again with OpenWRT and take the latest changes from @qosmio to make the 6.14.11 backports compatible.

Regular: https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-nsswifi-2025-07-08-2202
Mesh build: https://github.com/AgustinLorenzo/openwrt/releases/tag/ipq807x-nsswifi-mesh-2025-07-08-2320

I close this ticket and publish in the normal ways.

Regards, Agustin

closed 2025-07-09 by AgustinLorenzo