AgustinLorenzo/openwrt #67: USB Devices not mounting

digi111, 2025-08-01:
why don't you enable support for usb devices, I wonder? How many times will you close tickets, pretending that the problem is solved, although you stupidly do not solve it and pretend that it does not exist?

sinancetinkaya, 2025-08-02:
You can't do this with his firmware because it needs some kernel modules, which are not available in his repo.
It's actually very simple to make. 
You need to build your own firmware with `kmod-usb-storage-uas` and `block-mount` packages included. 
Add these line to `.config` file
```
# USB storage support
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-usb-storage-uas=y

# Block device support
CONFIG_PACKAGE_block-mount=y

# file system support
CONFIG_PACKAGE_kmod-fs-exfat=y
CONFIG_PACKAGE_kmod-fs-ntfs3=y
CONFIG_PACKAGE_kmod-fs-ext4=y
CONFIG_PACKAGE_kmod-fs-xfs=y

# samba server
CONFIG_PACKAGE_luci-app-ksmbd=y
```

trackhacs, 2025-08-08:
yeah its frustrating, but to be fair he is probably only compiling for himself and offers the result for puplic use.  its open source - so no obligation to any one - and this tone is not appropriate.

i also managed to use his source to compile my on variant (little bit tricky on arch though), currently running NSS, USB, SAMBA and wireguard. (still fighting with wguard - buts thats different story). 

i just changed config and followed the instructions here:

https://github.com/asvio/nbg7815-nss

sinancetinkaya, 2025-08-08:
He has to compile all the kernel modules, which will take quite some time, and he can't use OpenWRT's kernel modules either due to the kernel magic number mismatch.
So those who want a custom build with specific packages (USB disk, samba etc), have to recompile the firmware

digi111, 2025-08-08:
> yeah its frustrating, but to be fair he is probably only compiling for himself and offers the result for puplic use. its open source - so no obligation to any one - and this tone is not appropriate.
> 

> https://github.com/asvio/nbg7815-nss
The problem is that if he compiled it "for himself", then firstly, there is no need to upload it, since you are not asked. And secondly, what for himself, if he compiles it at the same time and for mx4300, and for rax120, and for ax9000.....he has been asked for several months, but this boy pretends not to understand what he is being asked for. Now another, more important question arises, why is this boy compiling firmware, for what purpose? What is he pursuing?

AgustinLorenzo, 2025-09-27:
> He has to compile all the kernel modules, which will take quite some time, and he can't use stable OpenWRT's kernel modules either due to the kernel magic number mismatch. So those who want a custom build with specific packages (USB disk, samba etc), have to recompile the firmware

A while ago, someone shared a workaround to avoid this very issue, but I can't find it right now.

Can you help me find it? I think that would fix the problem, right?

Snuupy, 2025-09-27:
> A while ago, someone shared a workaround to avoid this very issue, but I can't find it right now.
> 
> Can you help me find it? I think that would fix the problem, right?
> 
> [@Snuupy](https://github.com/Snuupy) Did he propose something?

Yes @arix00 made a change here: https://github.com/arix00/openwrt-mx4300/blob/d557ad5ec13b9e74caaea2b642674bc0c3afab6f/include/feeds.mk#L55 which adds kmods as a package in releases. You would still have to build it (once), but then everyone will be able to use it. The instructions to use it are here: https://github.com/arix00/openwrt-mx4300/blob/doc/nss-kmods.md

Please add this! 😅  @AgustinLorenzo

Snuupy, 2025-12-19:
Updated to latest, am able to install kmods, did not actually test the installed kmods yet