AgustinLorenzo/openwrt #68: Only 100MB ethernet after reboot with 2025-09-01-0221 on the XIAOMI AX3600

jenshenk, 2025-09-07:
Hi,
Noticed an odd behavior on my Xiaomi AX3600  (I'm new to the NSS enabled firmware and can only say it also happened with the version from 2-25-08-01): after a reboot I see on my switch that the AP tries twice  to connect at 1ghz but then falls back to 100mbit. After it is fully booted and I disconnect/reconnect the cable it chooses 1ghz and stays at it. Oddly enough the reboot after a recent firmware update it chooses 1ghz and stays there. 

Some context: Obviously the first thought might be a cabling issue, but I have CAT7 cabling and no other device shows this behavior at this specific LAN connection. Also the exact same device with the vanilla openwrt image never behaved like this after reboot. So my hunch is  something during booting might cause this and the flow reboot - 100mb - disconnect/reconnect cable - stable 1ghz is reproducible (tried it 5 times).   
 
Besides this small issue it's running very stable and it's not the end of the world as I'm very happy about the issues this firmware DOES resolve with the AX3600. But it would be cool for remote restarts if this could be fixed.

AgustinLorenzo, 2025-09-27:
Which brand is the other AP?

AgustinLorenzo, 2025-10-06:
Hi,

"unmanaged netgear switch".

That's a black box, hahahaha

Regards, Agustin